arduino-libraries/MIDIUSB #2: Can't get this working in Windows

stefandz, 2015-08-19:
I just end up with a composite device in Device Manager that has ```This device cannot start. (Code 10)``` as its status. Board is Leonardo, Arduino Build is 1.6.6 2015/08/19 10:30, code is MIDIUSB_test.ino.

stefandz, 2015-08-19:
Hi @facchinm - happy to help with anything I can at this end. Nice to see the commits resulting in a generic composite device now :+1: Can't wait for that to hit the mainstream!

facchinm, 2015-09-01:
Hi @stefandz , I had tested the library on XP and it was working so I thought everything was ok even on 7... I obviously was wrong :smile: 
As a workaround you could modify two lines in the core 

``` patch
diff --git a/hardware/arduino/avr/cores/arduino/USBCore.cpp b/hardware/arduino/avr/cores/arduino/USBCore.cpp
index f12d326..8374862 100644
--- a/hardware/arduino/avr/cores/arduino/USBCore.cpp
+++ b/hardware/arduino/avr/cores/arduino/USBCore.cpp
@@ -68,10 +68,10 @@ const u8 STRING_MANUFACTURER[] PROGMEM = USB_MANUFACTURER;
 #define DEVICE_CLASS 0x02
 
 //     DEVICE DESCRIPTOR
-const DeviceDescriptor USB_DeviceDescriptor =
+const DeviceDescriptor USB_DeviceDescriptorB =
        D_DEVICE(0x00,0x00,0x00,64,USB_VID,USB_PID,0x100,IMANUFACTURER,IPRODUCT,0,1);
 
-const DeviceDescriptor USB_DeviceDescriptorB =
+const DeviceDescriptor USB_DeviceDescriptor =
        D_DEVICE(0xEF,0x02,0x01,64,USB_VID,USB_PID,0x100,IMANUFACTURER,IPRODUCT,0,1);
 
 //==================================================================
```

I believe the bug is due to how Windows Vista and successors handle USB descriptors (much more compliant with standards than XP, Linux and OSX). 
The patch is not so safe so it would require a lot of testing before merging, however if you could try it and report the results it would be great!

NicoHood, 2015-09-01:
Thats what I also said about HID. the descriptor is technically wrong. It has to be different if its an HID device + Serial, Serial only etc. But its hard to add this to a pluggable USB-Core, I am aware of that. But this can cause you some trouble on several PCs. I dont know if you followed the Windows 10 issue (that got solved with the pluggable HID for some reason) but this could have been a similar issue. I still dont know why Pluggable HID solved the Windows 10 bug though.

stefandz, 2015-09-09:
I shall get on to testing this when I have some time, hopefully next week. @NicoHood - is it not possible to just use a 0x00, 0x00, 0x00, ... device descriptor as a general composite device descriptor no matter what the actual implemented interfaces are, leaving the OS to take a look at the interface descriptors and assign drivers appropriately? Also, what (if anything) is the difference between a 0x00, 0x00, 0x00, ... descriptor and a 0xEF, 0x02, 0x01, ... descriptor? As far as I understand, they are equivalent - but I feel I must be wrong!

NicoHood, 2015-09-09:
Normally zeros work, yes. But this **could** cause some issues on **some** Os. It is just an idea that you can test.

agdl, 2016-01-29:
Hi all after the PR made by @sandeepmistry that I merged it works good on windows

https://github.com/arduino-libraries/MIDIUSB/pull/10

Can you please try now? I also released it in the library manager wit the tag 1.0.2

facchinm, 2016-05-19:
Should be working now (all core have been updated accordingly)

closed 2016-05-19 by facchinm