arduino-libraries/MIDIUSB #3: MIDIUSB Examples Won't Compile

hyshka, 2015-11-04:
Hello,

I'm using the latest 64-bit Arduino 1.6.6 and in all of the example sketches I get the error:

    'midiEventPacket_t' was not declared in this scope

I'm running the 4.2.5 kernel on Arch Linux. Please advise if you have any tips for troubleshooting, I'm not sure what else to do.

facchinm, 2015-11-04:
Hi @hyshka , that's quite strange since `midiEventPacket_t` is defined in MIDIUSB.h
What sketch are you trying to compile?

hyshka, 2015-11-04:
Yes, I seen that in the source code, so I am unsure what's happening.

I've tried compiling MIDIUSB_Loop, MIDIUSB_Write, and MIDIUSB_Read.

Here's an attempt with full compile debugging:

```
/home/hyshka/arduino-1.6.6/arduino-builder -dump-prefs -logger=machine -hardware "/home/hyshka/arduino-1.6.6/hardware" -hardware "/home/hyshka/.arduino15/packages" -tools "/home/hyshka/arduino-1.6.6/tools-builder" -tools "/home/hyshka/arduino-1.6.6/hardware/tools/avr" -tools "/home/hyshka/.arduino15/packages" -built-in-libraries "/home/hyshka/arduino-1.6.6/libraries" -libraries "/home/hyshka/Arduino/libraries" -fqbn=arduino:avr:nano:cpu=atmega328 -ide-version=10606 -build-path "/tmp/build058231dadaf9a59d7b21f7d4d8680dec.tmp" -warnings=all -prefs=build.warn_data_percentage=75 -verbose "/home/hyshka/Arduino/libraries/MIDIUSB/examples/MIDIUSB_loop/MIDIUSB_loop.ino"
/home/hyshka/arduino-1.6.6/arduino-builder -compile -logger=machine -hardware "/home/hyshka/arduino-1.6.6/hardware" -hardware "/home/hyshka/.arduino15/packages" -tools "/home/hyshka/arduino-1.6.6/tools-builder" -tools "/home/hyshka/arduino-1.6.6/hardware/tools/avr" -tools "/home/hyshka/.arduino15/packages" -built-in-libraries "/home/hyshka/arduino-1.6.6/libraries" -libraries "/home/hyshka/Arduino/libraries" -fqbn=arduino:avr:nano:cpu=atmega328 -ide-version=10606 -build-path "/tmp/build058231dadaf9a59d7b21f7d4d8680dec.tmp" -warnings=all -prefs=build.warn_data_percentage=75 -verbose "/home/hyshka/Arduino/libraries/MIDIUSB/examples/MIDIUSB_loop/MIDIUSB_loop.ino"
"/home/hyshka/arduino-1.6.6/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -M -MG -MP -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR   "-I/home/hyshka/arduino-1.6.6/hardware/arduino/avr/cores/arduino" "-I/home/hyshka/arduino-1.6.6/hardware/arduino/avr/variants/eightanaloginputs" "/tmp/build058231dadaf9a59d7b21f7d4d8680dec.tmp/sketch/MIDIUSB_loop.ino.cpp"
"/home/hyshka/arduino-1.6.6/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -M -MG -MP -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR   "-I/home/hyshka/arduino-1.6.6/hardware/arduino/avr/cores/arduino" "-I/home/hyshka/arduino-1.6.6/hardware/arduino/avr/variants/eightanaloginputs" "-I/home/hyshka/Arduino/libraries/MIDIUSB/src" "/tmp/build058231dadaf9a59d7b21f7d4d8680dec.tmp/sketch/MIDIUSB_loop.ino.cpp"
"/home/hyshka/arduino-1.6.6/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -M -MG -MP -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR   "-I/home/hyshka/arduino-1.6.6/hardware/arduino/avr/cores/arduino" "-I/home/hyshka/arduino-1.6.6/hardware/arduino/avr/variants/eightanaloginputs" "-I/home/hyshka/Arduino/libraries/MIDIUSB/src" "/tmp/build058231dadaf9a59d7b21f7d4d8680dec.tmp/sketch/MIDIUSB_loop.ino.cpp"
"/home/hyshka/arduino-1.6.6/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -M -MG -MP -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR   "-I/home/hyshka/arduino-1.6.6/hardware/arduino/avr/cores/arduino" "-I/home/hyshka/arduino-1.6.6/hardware/arduino/avr/variants/eightanaloginputs" "-I/home/hyshka/Arduino/libraries/MIDIUSB/src" "/tmp/build058231dadaf9a59d7b21f7d4d8680dec.tmp/sketch/MIDIUSB_loop.ino.cpp"
"/home/hyshka/arduino-1.6.6/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -M -MG -MP -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR   "-I/home/hyshka/arduino-1.6.6/hardware/arduino/avr/cores/arduino" "-I/home/hyshka/arduino-1.6.6/hardware/arduino/avr/variants/eightanaloginputs" "-I/home/hyshka/Arduino/libraries/MIDIUSB/src" "/home/hyshka/Arduino/libraries/MIDIUSB/src/MIDIUSB.cpp"
"/home/hyshka/arduino-1.6.6/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -M -MG -MP -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR   "-I/home/hyshka/arduino-1.6.6/hardware/arduino/avr/cores/arduino" "-I/home/hyshka/arduino-1.6.6/hardware/arduino/avr/variants/eightanaloginputs" "-I/home/hyshka/Arduino/libraries/MIDIUSB/src" "/tmp/build058231dadaf9a59d7b21f7d4d8680dec.tmp/sketch/MIDIUSB_loop.ino.cpp"
"/home/hyshka/arduino-1.6.6/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR   "-I/home/hyshka/arduino-1.6.6/hardware/arduino/avr/cores/arduino" "-I/home/hyshka/arduino-1.6.6/hardware/arduino/avr/variants/eightanaloginputs" "-I/home/hyshka/Arduino/libraries/MIDIUSB/src" "/tmp/build058231dadaf9a59d7b21f7d4d8680dec.tmp/sketch/MIDIUSB_loop.ino.cpp"
"/home/hyshka/arduino-1.6.6/tools-builder/ctags/5.8-arduino2/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns "/tmp/build058231dadaf9a59d7b21f7d4d8680dec.tmp/preproc/ctags_target.cpp"
[... cut ...]
   midiEventPacket_t rx;
                     ^
MIDIUSB_loop:47: error: 'rx' was not declared in this scope
     rx = MidiUSB.read();
     ^
MIDIUSB_loop:47: error: 'MidiUSB' was not declared in this scope
     rx = MidiUSB.read();
          ^
MIDIUSB_loop:53: error: 'rx' was not declared in this scope
   } while (rx.header != 0);
            ^
Using library MIDIUSB at version 1.0.0 in folder: /home/hyshka/Arduino/libraries/MIDIUSB 
exit status 1
'midiEventPacket_t' was not declared in this scope
```

facchinm, 2015-11-04:
Ok, you are trying to compile for an Arduino nano with an ATMEGA328 which doesn't have a builtin USB capabilities, so the library won't work. 
Supported platform are 32u4 (Leonardo, Micro), SAM (Due) and SAMD (Zero),
You can still use normal MIDI library and use the embedded serial-to-usb chip to emulate a MIDI device

closed 2015-11-04 by facchinm

NicoHood, 2015-11-05:
I suggest to add this:
https://github.com/NicoHood/HID/blob/1a676a2d203bfa158f550fc9d6463aa533b1c3ec/src/HID-Project.h#L32-L38

Because we do not have the option to use things like that:
https://github.com/arduino/Arduino/pull/3982

facchinm, 2015-11-05:
@NicoHood , :+1:  I'm adding it right now