arduino-libraries/MIDIUSB #4: Sketch freezes until MIDI port is not opened by PC

odbol, 2015-11-08:
So I added this issue as [a comment to Arduino core](https://github.com/arduino/Arduino/pull/2943#issuecomment-154633457), but putting here for visibility (and ultimately this is where we'd apply the fix).

The whole Arduino sketch will freeze if you try using it after the USB is connected, but before an app opens the MIDI port and starts listening to the MIDI sent from the Arduino. 

Try with this test sketch: https://github.com/odbol/Arduino-MIDI-Test-Sketch

I've narrowed it down to this:

* When you plug the USB into a computer, the SerialUSB port initializes.
* The sketch calls MidiUSB.sendMIDI(), which freezes the whole sketch. (LED stops blinking)
* The reason, is that the USBD_Send() function blocks until there is a connection.
* Once you open a MIDI app on the PC (such as MIDI Monitor or Garageband), it opens the MIDI port, and the sketch unfreezes ( USBD_Send() stops blocking).
* Sending over SerialUSB works: the SerialUSB.write() function does not block if there is nothing connected. See the "if (_usbLineInfo.lineState > 0)" line in CDC.cpp for how they do this.
* We need to figure out a way to tell if USBD_Send() will block (i.e. the MIDI port is opened on the other end), and avoid calling it if it will block (just drop the messages if there's nothing listening).

I've attempted to duplicate the "_usbLineInfo.lineState" technique in CDC.cpp, but couldn't get it to work:

https://github.com/odbol/Arduino/tree/midi-fixserial

Any advice or help would be much appreciated!

odbol, 2015-11-08:
Tested on Mac OS X 10.10.5, Arduino 1.6.2.

facchinm, 2015-11-09:
Hi @odbol , the code you are trying to use is outdated, since in the meantime we went through a very big effort to support modular USB Core and Pluggable libraries.
Please test your sketch on IDE 1.6.6 with this library installed. 
Anyway you are right, Due and Zero will block because their `USB_Send()` has no implicit timeout on these architectures, so if there isn't a program on the host side sinking the bytes it will hang there forever. I'm sorry I can't help you more, since the "problem" is very core-centric and changing the implementation for `USB_Send()` is not feasible at the moment. Try asking the forum if someone else is interested in patching the usb core to perform non blocking operations and I'll be glad to spend some time on it :wink:

odbol, 2015-11-09:
@facchinm I think I fixed it now. BTW, thanks for the work on the PluggableUSB! Clever design, and such an improvement!

commit b360489c (2015-11-09): Fixed write() freezing entire Sketch unless MIDI port is opened by PC

multiple architecture version
fixes #4
closed 2015-11-09 by facchinm