arduino-libraries/MIDIUSB #13: 14bit control change messages

philmanofsky, 2016-02-15:
This really isn't an issue as much as it is a request, but i wasn't sure where to make requests. Anyways, could you write a function such as "controlChange14bit()" to support MIDI's 14-bit part of the protocol?

philmanofsky, 2016-02-27:
So I figured out how to do this. At first, I thought that ALL midi usb packets were 4 bytes. So I tried sending the CC MSB packet followed immediately by the CC's LSB packet. but that wasn't working. So I pulled out my USB Protocol Analyzer and hooked it up to a Behringer BCR-2000 which supports 14-bit MIDI and read the incoming packets. the 14-bit control change messages were actually combined into one 64-bit packet. So all you have to do to add this functionality to your library is to add something like this:

void MIDI_::send14bit(midiEventPacket_t eventMSB, midiEventPacket_t eventLSB)
{
	uint8_t data[8];
	data[0] = eventMSB.header;
	data[1] = eventMSB.byte1;
	data[2] = eventMSB.byte2;
	data[3] = eventMSB.byte3;
	data[4] = eventLSB.header;
	data[5] = eventLSB.byte1;
	data[6] = eventLSB.byte2;
	data[7] = eventLSB.byte3;
	write(data, 8);
}

fab672000, 2017-07-09:
Not sure it would be worth the add as this would be for me just a case where you could develop this same inline function that relies on  the write() method (which is public) so you can use it in your code without changing the MIDIUSB lib.

Also, something I don't understand is why you would need your workaround to make it work, so why it would not accept simply two packets to be sent this way (did not compile it but just for the example) :
```
MIDIUSB.sendMIDI(eventMSB);
MIDIUSB.sendMIDI(eventLSB);
MIDIUSB.flush(); // this should ensure we send only one USB midi message for both events
```