arduino-libraries/MIDIUSB #19: Sending arbitrary length MIDI messages

vockleya, 2016-06-08:
I'm working on a project that requires me to send a different length of MIDI message than the standard 4-byte note on/off, and program/control change messages. It seems like this should be trivial, but I'm having trouble getting it to work. It looks like I should be able to use the MidiUSB.write() function to send arbitrary data to the computer, but I'm getting very odd messages on the computer. Depending on the content of the message, I get anywhere from nothing, to a single byte, to something random that wasn't even in the original message.

Here's the snippet of code that is actually sending the data.
> uint8_t data[7] = {0xF0, 0x7F, 0x7F, 0x02, 0x10, 0x01, 0xF7};
> MidiUSB.write(data, 7);
> MidiUSB.flush();

The message I'm trying to send is a SysEx MIDI show control "go" command (and various others once I get this working). According to the SysEx/MSC spec, I'm using the correct byte sequence.

As far as I can tell, all the sendMIDI() function does is create a data buffer from the midiEventPacket_t passed to it, and then calls write(). I'm just creating my own buffer and directly calling write(). It's interesting to note that when I use a standard 4-byte note on message in my code above, it works correctly.

monteslu, 2016-10-06:
Just hit this same thing.  Trying to use https://github.com/firmata/protocol/blob/master/protocol.md on top of MIDIUSB but need variable length messages as well.

oqibidipo, 2016-10-07:
USB MIDI event packets are always 4 bytes long with a header byte and up to 3 bytes of MIDI data (padded with zeros if necessary). You have to split the SysEx message into several packets.

Untested code:
```
void MidiUSB_sendSysEx(const uint8_t *data, size_t size)
{
    midiEventPacket_t event;
    const uint8_t *d = data;
    size_t bytesRemaining = size;
    
    while (bytesRemaining > 0) {
        switch (bytesRemaining) {
        case 1:
            event.header = 5;   // SysEx ends with following single byte
            event.byte1 = *d;
            event.byte2 = 0;
            event.byte3 = 0;
            bytesRemaining = 0;
            break;
        case 2:
            event.header = 6;   // SysEx ends with following two bytes
            event.byte1 = *d++;
            event.byte2 = *d;
            event.byte3 = 0;
            bytesRemaining = 0;
            break;
        case 3:
            event.header = 7;   // SysEx ends with following three bytes
            event.byte1 = *d++;
            event.byte2 = *d++;
            event.byte3 = *d;
            bytesRemaining = 0;
            break;
        default:
            event.header = 4;   // SysEx starts or continues
            event.byte1 = *d++;
            event.byte2 = *d++;
            event.byte3 = *d++;
            bytesRemaining -= 3;
            break;
        }
        MidiUSB.sendMIDI(event);
    }
}
```

oqibidipo, 2016-10-07:
Results of some testing with MIDI Monitor & SysEx Librarian on OS X 10.11.6:
* seems to work better with a small delay (1 ms is enough, maybe even less) between the packets, without the delay I get different sized messages although the sent message is always the same
```
        MidiUSB.sendMIDI(event);
        if (bytesRemaining > 0) delay(1);
```
* 2 kernel panics :(

oqibidipo, 2016-10-07:
This version sends all packets at once, no delay needed.
```
void MidiUSB_sendSysEx(const uint8_t *data, size_t size)
{
    if (data == NULL || size == 0) return;

    size_t midiDataSize = (size+2)/3*4;
    uint8_t midiData[midiDataSize];
    const uint8_t *d = data;
    uint8_t *p = midiData;
    size_t bytesRemaining = size;
    
    while (bytesRemaining > 0) {
        switch (bytesRemaining) {
        case 1:
            *p++ = 5;   // SysEx ends with following single byte
            *p++ = *d;
            *p++ = 0;
            *p = 0;
            bytesRemaining = 0;
            break;
        case 2:
            *p++ = 6;   // SysEx ends with following two bytes
            *p++ = *d++;
            *p++ = *d;
            *p = 0;
            bytesRemaining = 0;
            break;
        case 3:
            *p++ = 7;   // SysEx ends with following three bytes
            *p++ = *d++;
            *p++ = *d++;
            *p = *d;
            bytesRemaining = 0;
            break;
        default:
            *p++ = 4;   // SysEx starts or continues
            *p++ = *d++;
            *p++ = *d++;
            *p++ = *d++;
            bytesRemaining -= 3;
            break;
        }
    }
    MidiUSB.write(midiData, midiDataSize);
}
```

monteslu, 2016-10-07:
This is great @oqibidipo !

Any idea how'd we go about reading SYSEX messages into the device?

thenetimp, 2018-01-05:
@monteslu did you ever figure out how to read sysex messages on your arduino?

monteslu, 2018-01-05:
@thenetimp I wish :)
I'd really love to get firmata running directly over midi to an arduino device.  I'm just not good enough at c++ to fix this library myself.

thenetimp, 2018-01-05:
I am working on a usb stomp box,  I'd love to be able to send sysex messages to map buttons to channels, but my c++ is very much shit.  *le sigh*

sanotronics, 2018-07-13:
Hi!

Anyone got this working?

I also need to be able to receive sysex on a SAMD21.

I used @oqibidipo's function to send sysex and it seems to work fine.

Maybe it is a matter of porting the [https://github.com/FortySevenEffects/arduino_midi_library/blob/a44d1e7c1c5b8129d8437f9003df2be8a916a67e/src/MIDI.hpp#L676](parse()) function of the Arduino MIDI Library to this library, what do you think?

franky47, 2018-11-06:
Conversation about integration with the MIDI Library (which now includes an interface with MIDIUSB and allows sending and receiving "large" SysEx messages) is here: https://github.com/arduino-libraries/MIDIUSB/issues/21#issuecomment-435800047

lathoub, 2020-10-04:
This issue is resolved in https://github.com/lathoub/Arduino-USBMIDI (fully supporting receiving and sending `SysEx`) that uses the underlaying [FortySevenEffects Arduino MIDI Library](https://github.com/FortySevenEffects/arduino_midi_library) (USB-MIDI is one of the transport mechanisms ). See also #21