arduino-libraries/MIDIUSB #22: Question: How can I change the USB requested power ?

rjkorn, 2016-11-26:
I noticed my micro (32u4) device reports it needs 500ma only when the library is used. That upsets one of my computers. Actual current use is 50-60ma for my circuit. Is there and easy way to redefine it?

when it enumerates as MIDI it says:
  Current Required (mA):	500

The initial  bootloader  says:
  Current Required (mA):	100

The defines I'd like to mimic are:

#define USB_CFG_IS_SELF_POWERED         0
/* Define this to 1 if the device has its own power supply. Set it to 0 if the
 * device is powered from the USB bus.
 */

#define USB_CFG_MAX_BUS_POWER           100
/* Set this variable to the maximum USB bus power consumption of your device.
 * The value is in milliamperes. [It will be divided by two since USB
 * communicates power requirements in units of 2 mA.]
 */

 Thank you for any insight....

facchinm, 2016-11-28:
Hi @rjkorn , the "culprit" is not the MIDIUSB library but the core itself. We define 500mA since we are exposing as an USB 2.1 and 500mA is the minimum requirement. You can easily lower the value by modifying this file https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/cores/arduino/USBCore.h#L270 (the file could be in `hardware` folder or in `$Arduino15/packages/hardware/arduino/` folder as well, depending on your installation)

rjkorn, 2016-11-29:
Perfect, thank you.

I couldn't find that one.....

closed 2016-11-29 by rjkorn

fab672000, 2016-11-30:
I also had this issue and did that patch when implementing my midi 1x1 interface for my iPad, 
ideally we probably should be able to test if this constant is defined and only define it if not ?
something like:

```
#if !defined(ARDUINO_MAX_POWER_MA)
# define ARDUINO_MAX_POWER_MA 500
#endif

#define D_CONFIG(_totalLength,_interfaces) \
	{ 9, 2, _totalLength,_interfaces, 1, 0, USB_CONFIG_BUS_POWERED | USB_CONFIG_REMOTE_WAKEUP, USB_CONFIG_POWER_MA(ARDUINO_MAX_POWER_MA) }

```

facchinm, 2016-11-30:
@fab672000 , I took a look at your MIDIUSB library and the 20mA "workaround" to let it work with an iPad. Your proposal is good but, since there is no way to pass a define to the core without modifying the board files (or adding a new board), I'd rather go with a plain library function (https://github.com/facchinm/Arduino/commit/8dd26f2ba9cf9f203981d6f19724429da3a109d0) . The overhead is quite low since `D_CONFIG` is already dynamic and maybe there are lots of use cases with OTG devices.
@cmaglie any thoughts on that?

cmaglie, 2016-11-30:
should you call this function before the descriptors are sent?

facchinm, 2016-11-30:
Adding it before `SerialUSB.begin()` in the sketch seems to be enough

fab672000, 2016-12-02:
> I'd rather go with a plain library function (facchinm/Arduino@8dd26f2) 
Nice!
I was initially thinking it was in ROM but after checking for its use ; it is store dynamically in RAM, so that should work fine, and this would be for me a good fix.
+1 on that proposal, the new method would deserve a doxygen comment on to how it should be used before SerialUSB.begin() IMHO,
Thanks @facchinm