arduino-libraries/MIDIUSB #26: How can I change the name of the device?

fraramon, 2017-01-26:
I would like to change the name "Arduino Leonardo".
I have not found any place to change it ..

facchinm, 2017-01-27:
Are you on Windows? In that case the name displayed comes from the driver and there is nothing we can do. Otherwise, modify this line https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/boards.txt#L298, recompile your project and name of the board will change accordingly :slightly_smiling_face:

closed 2017-01-27 by facchinm

oqibidipo, 2017-01-27:
If you want to switch between "Leonardo" and custom names quickly, you can add a submenu under Tools by adding lines like these to `boards.txt`:
```
menu.usb_name=USB Product Name

leonardo.menu.usb_name.leonardo="Arduino Leonardo"
leonardo.menu.usb_name.leonardo.build.usb_product="Arduino Leonardo"

leonardo.menu.usb_name.leonardo_1="Leonardo #1"
leonardo.menu.usb_name.leonardo_1.build.usb_product="Arduino Leonardo #1"

leonardo.menu.usb_name.leonardo_2="Leonardo #2"
leonardo.menu.usb_name.leonardo_2.build.usb_product="Arduino Leonardo #2"
```