arduino-libraries/MIDIUSB #33: Error compiling and uploading to Mega or Uno

BLCKPSTV, 2017-04-12:
When trying to upload this code I get following error..

Arduino: 1.8.2 (Mac OS X), Board:"Arduino/Genuino Uno"

In file included from /Users/myname/Downloads/MIDI_Controller/MIDI_Controller.ino:2:0:
/Users/myname/Documents/Arduino/libraries/MIDIUSB/src/MIDIUSB.h:16:2: error: #error MIDIUSB can only be used with an USB MCU.
 #error MIDIUSB can only be used with an USB MCU.
  ^
exit status 1
Fout bij het compileren van board Arduino/Genuino Uno

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Franky47, 2017-04-13:
The error is quite explicit: the library cannot be used with non-native USB boards, and neither Mega nor Uno have native USB capabilities.
If you want a USB native compatible board, look for an Arduino Leonardo (or any other ATmega32u4-based board).

fab672000, 2017-04-14:
MIDIUSB is also available for arduino ARM SAM cores like in Arduino Due and Arduino Zero boards (SAM core > 1.6.4, SAMD core > 1.6.1)

closed 2017-04-14 by cmaglie