arduino-libraries/MIDIUSB #34: arduino not recognized as midi device

phil123456, 2017-04-30:
Hi,

I wrote a simple prgram that ramps up and down but apparently arduino doe snot get recognized as a midi device

how do I send midi infos to tell my DAO that the arduino is a midi controller ?

`#include "MIDIUSB.h"

void setup() {
  Serial.begin(115200);
}

void controlChange(byte channel, byte control, byte value) {
  midiEventPacket_t event = {0x0B, 0xB0 | channel, control, value};
  MidiUSB.sendMIDI(event);
}

void loop() 
{
  for(int i=0;i<255;i++)
  {
    controlChange(0, 10, i); // Set the value of controller 10 on channel 0 to 65
    delay(10);
  }
  
  for(int i=255;i<0;i--)
  {
    controlChange(0, 10, i); // Set the value of controller 10 on channel 0 to 65
    delay(10);
  }
}`

thanks

phil123456, 2017-04-30:
ok it works, the issue was in sonar

facchinm, 2017-05-02:
:+1: closing the issue, reopen if needed.

closed 2017-05-02 by facchinm