arduino-libraries/MIDIUSB #36: Read doesn't seem to work with Arduino Due

aethersis, 2017-06-11:
Regardless of what I try, read doesn't read anything from USB MIDI keyboard. 

The setup is as follows:
-CME UF-80 MIDI keyboard plugged into Native USB port in Arduino Due (works on PC) using USB to micro USB host adapter. The adapter was tested on the same Arduino Due with mouse and it works.
-12V power supply plugged into Arduino Due, Separate 12V power supply plugged into CME UF-80 (Also tried without PSUs and with 1 PSU on either end)
-Running read_example from this library

Nothing is being sent by Serial.write(). The function works normally but nothing seems to be received by MIDIUSB::read(). 

What could be the possible source of the issue?

facchinm, 2017-06-12:
Hi @aethersis ,
sorry for the confusion but this library is intended to act as an USB MIDI device, not as the host. To achieve what you want to do you need a patched USBHost library with MIDI messaging reading capabilities.

closed 2017-06-12 by facchinm

aethersis, 2017-06-12:
Do you know of any library having this capability?

facchinm, 2017-06-12:
[This](https://github.com/YuuichiAkagawa/USBH_MIDI) looks exactly what you need but it applies to USBHost2 library (which, AFAIR, only had USBHost shield as target but I could be wrong).