arduino-libraries/MIDIUSB #37: How to change de MIDI-USB device name.

jtragtenberg, 2017-06-13:
Hello, How can I change de MIDI-USB device name?  It only appears as the boards name, "Arduino Leonardo" or "Arduino Micro". I want it to be the name of my controller, and also change the Descriptor Product name, manufacturer and model names as well as the device and manufacturer ID number?

Thank you

facchinm, 2017-06-14:
Hi @jtragtenberg ,
the easiest way is creating a new "core" derived by one of ours and define your custom board this way (taking a Leonardo as example)
```
myboard.build.mcu=atmega32u4
myboard.build.f_cpu=16000000L
myboard.build.vid=BOARD_VID
myboard.build.pid=BOARD_PID
myboard.build.usb_product="BOARD_NAME"
myboard.build.usb_manufacturer="MANUFACTURER_NAME"
myboard.build.board=AVR_LEONARDO
myboard.build.core=arduino:arduino
myboard.build.variant=arduino:leonardo
myboard.build.extra_flags={build.usb_flags}
```

This way you can customize the values without touching the core files.

closed 2017-06-14 by facchinm

exabrial, 2017-10-31:
Sorry to ask a stupid question, but where do those configuration entries go? I tried putting those exact values in `/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/boards.txt` on OSX and did not see a new board drop down in the Arduino IDE. Thanks!

facchinm, 2017-10-31:
@exabrial the easiest way is to create a new `hardware` folder in `$sketchbook/hardware/fancyname/$architecturename` with a `boards.txt` file in there.
You sketchbook depends on the OS but it's `/Users/yourname/Documents/Arduino` on OSX.
`architecturename` must match the one you are expanding (eg. `avr` or `samd`). `fancyname` can be anything that doesn't override the default one (so no `arduino`).
After creating this folder you should restart the IDE to see the new boards in the menu. To avoid starting from scratch you can use https://github.com/webusb/arduino/tree/gh-pages/hardware/webusb/avr as a reference

eqcq, 2017-11-08:
HI and thx for that. I've followed all the steps, I find my device in arduino IDE but when I compile I read that : recipe.preproc.macros est un schéma manquant

Have you any idea ?

THX

musinou, 2018-01-20:
Hi everybody, 
After reading this thread (thank you!) and a few other, I ended up making a little how-to on there, hope it may be some help for someone. 

http://liveelectronics.musinou.net/MIDIdeviceName.php

eqcq, 2018-01-20:
Hi thanks a lot for that.

musinou, 2018-01-20:
No problem!

kalandara, 2018-03-25:
http://liveelectronics.musinou.net/MIDIdeviceName.php

Here is description.
It's working I tried everything is FINE

musinou, 2018-03-25:
Good, thank you for the feedback!

eltinello, 2020-03-27:
Old Post, but maybe someone will need it:
On Windows devices same procedure is possible but tha location of the relevant files is different

You'll find the file in the following (hidden) folder:
C:\Users\$YOURUSERNAME$\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.2.1
Of course for different processors the path may vary a bit.

Filename --> board.txt
Varname --> xxx.build.usb_product i.e. adafruit_feather_m0.build.usb_product="Feather M0"

Important notes:
- restart Arduino IDE after modification.
- All flashed devices will have the new name. Consider undoing the change after flashing your sketch onto arduino.

Procedure worked for Adafruit Feather M0 and Trellis M4.

HTH