arduino-libraries/MIDIUSB #38: compilation error

vjdobermann, 2017-07-09:
Hi team,

I try to use this library but when I want to upload my code to my arduino uno, I get this error message 👍 

In file included from /Users/vjdobermann/Documents/Arduino/push_button_midi/push_button_midi.ino:2:0:
/Users/vjdobermann/Documents/Arduino/libraries/MIDIUSB-master/src/MIDIUSB.h:18:2: error: #error MIDIUSB can only be used with an USB MCU.
 #error MIDIUSB can only be used with an USB MCU.
  ^
exit status 1
Erreur de compilation pour la carte Arduino/Genuino Uno

Could someone help me please ?

fab672000, 2017-07-09:
Your MCU for the Uno is an atmega328p which is not a USB MCU. So as it states ; it won't work because MIDIUSB needs a USB MCU like on the arduino: leonardo, (pro) micro, Zero or Due.

vjdobermann, 2017-07-09:
Hi fab,
thanks for you answer.
In my project I need to use a midi signal, is still possible to send midi over usb câble with the Uno ?

fab672000, 2017-07-09:
Yes it is possible,  but you need a different _software_  usb library with a dedicated bootstrap like [hiduino ](https://github.com/ddiakopoulos/hiduino) (you will need to upload the code with a programmer).

vjdobermann, 2017-07-09:
oh does it exist a solution more simple ?
Or it's definitely impossible to send out midi signal from an uno ?

per1234, 2017-07-09:
No it's not impossible. @fab672000 already explained how it can be done. It's not simple. Just buy an Arduino Leonardo or Pro Micro.

vjdobermann, 2017-07-13:
finally I've founded a solution with this [http://projectgus.github.io/hairless-midiserial/#getting_started](http://projectgus.github.io/hairless-midiserial/#getting_started)

facchinm, 2017-07-18:
It's a sort of workaround but it definitely works ;) I'm closing the issue as solved, thanks!

closed 2017-07-18 by facchinm