arduino-libraries/MIDIUSB #39: Can´t Assign Potentiometer to volume-effects pedal

cferrarini, 2017-07-21:
The potentiometer is related only to note intensity, cannot be assigned to a wahwah or volume pedal alone,
it wold be nice to add a proper command in the library o transmit continuously the pot state through midi commands on usb. The software Im using is guitar rig. The sketch in the example compiled perfectly on arduino micro, except for the notes line c3,d3 etc, in this line I changed to 48, 49, 50 to compile.

Thanks, Celso

fab672000, 2017-07-21:
Hi Celso, your question is not clear please explain what you would to achieve first. You seem to mix your own compilation problems with something not working. It seems you want to build some sort of pedalboard switch for your guitar software plugin to control in your DAW application ? It should work fine in you use midi cc. You try to use notes which is not the proper approach for midi controls.
Check[Summary of midi message]( https://www.midi.org/specifications/item/table-1-summary-of-midi-message)

cferrarini, 2017-07-21:
Exactly, With the provided sketch "https://www.arduino.cc/en/Tutorial/MidiDevice" I was able to play the mapped notes, with the pot acting as intensity of the note, which is packaged and sent as a Note on/Note off command. Exactly what it is intended for. 
The Guitar Pro Software Has a Learn Button to capture the notes and assign to functions, such as enable/disable custom effects. This already works fine.
One thing I noticed is that when the pot is set to minimum the buttons do not work, so the software is recording the note intensity  value learned.
What I want to achieve is to make the potentiometer to control the volume slider or wah wah pedal, the software  learns when a button is pressed but just goes on or off, the note intensity pot does not drive the gradative slider. The provided sketch example does not include a proper control or that. Does the library provide  a specific command to midi gradative control?
Thanks for your attention.
Celso

fab672000, 2017-07-21:
I see, you have to develop a program (your starting example is finethat convert analog input (i.e. your pot) to a range that can be sentvia the  controlChange() function, not the note on/off functions. The value you would read from say, the analog arduino port A0, would be sent via this function using the 'value' parameter that now would be your analog input value.
This is not a MIDIUSB issue,please continue this discussion in arduin o forum where you will find a lot of people to help on your project design. See [Arduino Project Guidance](https://forum.arduino.cc/index.php?board=3.0)

facchinm, 2017-07-21:
Since it's not a bug in the library, I'd close the issue and continue the discussion on the [forum](https://forum.arduino.cc) . Thanks!

closed 2017-07-21 by facchinm