arduino-libraries/MIDIUSB #40: Avoid long time execution when sending midiusb packet

YruamaLairba, 2017-08-19:
I just discovered that sometimes, sending a midiusb packet take more than 250ms. Typically this happen when the Arduino is connected to the computer, but no app to get the midi data. Is there a simple way to avoid or reduce this delay from the arduino side ?

facchinm, 2017-09-04:
Hi @YruamaLairba ,
probably what you are seeing is the `USB_send` timeout if the interface is not yet configured.
Do you also experience the long delay if you open the serial port?

YruamaLairba, 2017-09-04:
Yes, the long delay is here when the serial port is open. This how i report the 250ms delay. The `USB_send` timeout seems a good way, can you tell me more about it or indicate some documentation ?

franky47, 2018-11-06:
Related to #46.

DisasterAreaDesigns, 2019-03-24:
On the Due, the library checks Is_udd_write_Enabled, which tests UOTGHS_DEVEPTISR[0] and UOTGHS_DEVEPTISR_RWALL.  The first value is the TXINI, Transmitted IN Data interrupt which is set when the current bank is ready to accept a new IN packet. The second is RWALL, Read Write ALLowed, which is set when the current bank is not full and has room for more bytes.

So in a nutshell, the Due will allow the user application to send USB when the current bank is ready to send and has room in its FIFO buffer.  As soon as either of those cases is not true, the USB MIDI library will skip packets until the FIFO is cleared or TXINI is ready to send.

All that stuff is on the SAM3X datasheet, starting around page 1059 under USB Interrupt System.  I've been trying to match them up to interrupts on the SAMD21 but it just doesn't have the same capabilities.

If you look at the USB MIDI library, there's that nice check for the Due version but if the architecture is SAMD, it just sets the is_write_enabled() to 1, which means it will _always_ fail if no host is listening.

The SAMD21 USB interrupt mapping starts on page 757 of its datasheet, and it lists out 10 flags for the device interrupts and 6 for each endpoint.  None of them seem to be helpful for determining whether the FIFO is full or the bank is ready to send.

Can we leverage EPINTFLAG.TRFAIL0 / TRFAIL1 to show a transmit fail?  Failing that, can we just flush the FIFO or whatever TX buffer we're using?  It seems that the Due will send until it fills the FIFO and then stop, so I assume that overflowing the FIFO is the problem?  If so, then can we just roll over the buffer on overflow if nobody is listening?  I really don't care about missing messages with no listener present, I can re-send them later.  What I _do_ care about is the whole application locking if the listener terminates or isn't present at USB insertion / enumeration.

DanielRudrich, 2019-10-04:
There's an easy way to check, if the port is ready. See my reply here: https://github.com/arduino-libraries/MIDIUSB/issues/46