arduino-libraries/MIDIUSB #42: Note board limitations in documentation

awgrover, 2017-10-25:
It looks like this library will only work for certain arduino boards. Maybe the same limitations as noted in https://www.arduino.cc/en/Reference/MouseKeyboard ? Adding a note to the README would be nice.

facchinm, 2017-10-26:
Hi @awgrover ,
the reference you mentioned specifies
```
These core libraries allow a 32u4 based boards or Due and Zero board to appear as a native Mouse and/or Keyboard to a connected computer.
```
It should be enough, although I think we just need a better periphrasis to describe boards (processors) with native USB capabilities.

facchinm, 2017-10-26:
Sorry, I totally misinterpreted the issue. I'm adding some info on README to make it super clear

closed 2017-10-26 by facchinm