arduino-libraries/MIDIUSB #43: Allow for multiple virtual MIDI cables (ports)

ab5tract, 2017-12-28:
The class compliant USB MIDI spec allows for 16 separate MIDI "virtual cables" (their terminology). This library appears to hardcode all transmissions to a single port.

snodecoder, 2020-09-02:
16 virtual cables can only be used if a Teensy board is used. According to the datasheet s the Leonardo / Pro Micro boards have 6 configurable USB endpoints, so besides the 2 endpoints in use for Serial via USB, it should be possible to configure it with a total of 4 virtual cables, right?

Does anyone know what needs te be changed to allow for multiple virtual cables? I would like to help in adding this functionality, but I don't know where to begin ;)

lathoub, 2020-10-04:
https://github.com/lathoub/Arduino-USBMIDI implements cables (and uses this library to implement the underlying USB calls).
See example https://github.com/lathoub/Arduino-USBMIDI/tree/master/examples/CableNumber (specific cable nr) and https://github.com/lathoub/Arduino-USBMIDI/blob/master/examples/SysEx_Teensy4.1/SysEx_Teensy4.1.ino (multiple cables)