arduino-libraries/MIDIUSB #48: Receivng Sysex and Sysex Raw MIDI messages in MIDIUSB

kaspencer, 2018-11-06:
uController=Arduino Due
LIbrary=MIDIUSB
I wish to receive Sysex and Sysex Raw MIDI messages using a Due. However I suspect that this is not possible as the read() function seems only to expect a structure of very few data bytes.
Can you confirm the MIDIUSB does indeed not support Sysex message reading?
Can you possibly indicate by which library how this may be done using an Arduino Due?
I am grateful for your work developing the library, and also will be for your help with this issue.
KASpencer

franky47, 2018-11-06:
The USB MIDI specification splits SysEx messages into packets, this library should allow to receive such split data, but you'd need a way to recompose the message. I have a pre-release of the Arduino MIDI library doing this work and interface this MIDIUSB library with your sketch. Check out the discussion in #21: 
https://github.com/arduino-libraries/MIDIUSB/issues/21#issuecomment-435800047

gdsports, 2018-11-06:
You can compose the 4 bytes raw USB MIDI packets into complete sysex messages.
```
  // init sysex buffer empty
  midiEventPacket_t rx = MidiUSB.read();
  if (rx.header != 0) {
    switch (rx.header & 0x0F) {
      case 0x00:  // Misc. Reserved for future extensions.
        break;
      case 0x01:  // Cable events. Reserved for future expansion.
        break;
      case 0x02:  // Two-byte System Common messages
      case 0x0C:  // Program Change
      case 0x0D:  // Channel Pressure
        // do something with 2 byte messages
        break;
      case 0x03:  // Three-byte System Common messages
      case 0x08:  // Note-off
      case 0x09:  // Note-on
      case 0x0A:  // Poly-KeyPress
      case 0x0B:  // Control Change
      case 0x0E:  // PitchBend Change
        // do something with 3 byte messages
        break;
      case 0x04:  // SysEx starts or continues
        // append sysex buffer with 3 bytes
        break;
      case 0x05:  // Single-byte System Common Message or SysEx ends with the following single byte
        // append sysex buffer with 1 byte
        // process completed sysex buffer
        // init sysex buffer empty
        break;
      case 0x06:  // SysEx ends with the following two bytes
        // append sysex buffer with 2 bytes
        // process completed sysex buffer
        // init sysex buffer empty
        break;
      case 0x07:  // SysEx ends with the following three bytes
        // append sysex buffer with 3 bytes
        // process completed sysex buffer
        // init sysex buffer empty
        break;
      case 0x0F:  // Single Byte, TuneRequest, Clock, Start, Continue, Stop, etc.
        // process since byte messages
        break;
    }
  }

```

kaspencer, 2018-11-06:
Thankyou Franky and gdsports for those helpful replies.
I will get on with testing those ideas ASAP. I will report back as seems necessary, but I hope that's enough for me to get on with the project.
Best wishes,
Kenneth Spencer

lathoub, 2020-10-04:
This issue is resolved in https://github.com/lathoub/Arduino-USBMIDI (fully supporting receiving and sending `SysEx`) that uses the underlaying [FortySevenEffects Arduino MIDI Library](https://github.com/FortySevenEffects/arduino_midi_library) (USB-MIDI is one of the transport mechanisms ).

kaspencer, 2020-10-04:
lathoub: thanks for that. I'm glad to say that the project was completed successfully this year, and made use of the librares which you mentioned.
Best wishes,
Kenneth Spencer