arduino-libraries/MIDIUSB #50: Is there an easy way to have MIDI only, without serial?

amowry, 2019-01-02:
This library is great! I was wondering if there's a way to to turn off serial in order to make fully class-compliant devices, similar to the way Teensy works? Thanks!

amowry, 2019-01-03:
Okay, thanks—I did end up finding a way to remove the CDC by modifying the core files (I’m using a standalone 32u4). Assuming CDC is removed, should the library create a class-compliant device?

facchinm, 2019-01-04:
This patch should be enough to get rid of CDC on 32u4 and obtain a compliant MIDIUSB device
```patch
--- a/cores/arduino/USBCore.cpp
+++ b/cores/arduino/USBCore.cpp
@@ -331,10 +333,6 @@ int USB_Send(uint8_t ep, const void* d, int len)
 uint16_t _initEndpoints[USB_ENDPOINTS] =
 {
        0,                      // Control Endpoint
-       
-       EP_TYPE_INTERRUPT_IN,   // CDC_ENDPOINT_ACM
-       EP_TYPE_BULK_OUT,       // CDC_ENDPOINT_OUT
-       EP_TYPE_BULK_IN,        // CDC_ENDPOINT_IN
 
        // Following endpoints are automatically initialized to 0
 };
@@ -382,9 +380,6 @@ bool ClassInterfaceRequest(USBSetup& setup)
 {
        uint8_t i = setup.wIndex;
 
-       if (CDC_ACM_INTERFACE == i)
-               return CDC_Setup(setup);
-
 #ifdef PLUGGABLE_USB_ENABLED
        return PluggableUSB().setup(setup);
 #endif
@@ -473,8 +468,6 @@ static uint8_t SendInterfaces()
 {
        uint8_t interfaces = 0;
 
-       CDC_GetInterface(&interfaces);
-
 #ifdef PLUGGABLE_USB_ENABLED
        PluggableUSB().getInterface(&interfaces);
 #endif
@@ -862,8 +855,8 @@ bool USBDevice_::wakeupHost()
        return false;
 }
 
-PluggableUSB_::PluggableUSB_() : lastIf(CDC_ACM_INTERFACE + CDC_INTERFACE_COUNT),
-                                 lastEp(CDC_FIRST_ENDPOINT + CDC_ENPOINT_COUNT),
+PluggableUSB_::PluggableUSB_() : lastIf(0),
+                                 lastEp(0),
                                  rootNode(NULL), totalEP(USB_ENDPOINTS)
 {
        // Empty
``` 

The resulting descriptor is
```
sudo lsusb -vvv -d2341:8036

Bus 001 Device 060: ID 2341:8036 Arduino SA Leonardo (CDC ACM, HID)
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0x2341 Arduino SA
  idProduct          0x8036 Leonardo (CDC ACM, HID)
  bcdDevice            1.00
  iManufacturer           1 Arduino LLC
  iProduct                2 Arduino Leonardo
  iSerial                 3 MIDI
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
[... 101 lines cut ...]
          bNumEmbMIDIJack         1
          baAssocJackID( 0)       1
      Endpoint Descriptor:
        bLength                 9
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
        bRefresh                0
        bSynchAddress           0
        MIDIStreaming Endpoint Descriptor:
          bLength                 5
          bDescriptorType        37
          bDescriptorSubtype      1 (GENERAL)
          bNumEmbMIDIJack         1
          baAssocJackID( 0)       3
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)
```

amowry, 2019-01-04:
Thank you! Is this for an earlier version of USBCore.cpp? I don't see these lines:

`-PluggableUSB_::PluggableUSB_() : lastIf(CDC_ACM_INTERFACE + CDC_INTERFACE_COUNT),
-                                 lastEp(CDC_FIRST_ENDPOINT + CDC_ENPOINT_COUNT),`

However, I made the other changes and it seems to have removed CDC. I'm not sure how to tell for sure if it did it correctly ;) For anyone who is interested, the file can be put in the sketch folder, so it will only affect the given sketch.