arduino-libraries/MIDIUSB #55: USB MIDI incompatible device

rodedski, 2019-04-05:
I need to be pointed in a direction to solve this issue.
I have programmed an Arduino Pro Micro using a midiusb example.
I have run the example with it attached to my Macbook and the MIDI Monitor is working as expected with the MIDI commands displayed perfectly.
The problem starts when I plug it into my Yamaha Tyros 3. The Arduino Pro Micro device is not recognized by the Yamaha which says that an unrecognized or incompatible device has been connected.
After researching I have scrambled my brain trying to figure out how I can re-program the Arduino with a different USB device signature. I don't even know how to find out what the device signature of a MIDI controller should be for it to be recognized by the Yamaha. I am certain this issue has been cracked before but its just a question of finding it. Help me please. Rod

facchinm, 2019-04-08:
Hi @rodedski ,
when you'll find out the accepted vid/pid combo that is accepted by your Yamaha keyboard you can change the Micro signatures here https://github.com/arduino/ArduinoCore-avr/blob/5755ddea49fa69d6c505c772ebee5af5078e2ebf/boards.txt#L404-L405
Anyway, I believe that if the Yamaha has a MIDI IN via standard midi connector it may be more convenient to use a proper MIDI library (you won't need any extra driver + it works for every device with at least one serial port) https://github.com/FortySevenEffects/arduino_midi_library/

rodedski, 2019-04-08:
Thank you for your help. I found the page and I will try to follow what it means. Since my post I have started to wonder if it is worth the effort and even if it even possible. At least you have confirmed that it is possible to adapt the board but I am not convinced that I would be able to work out the method. You have at least inspired me and given me fresh hope.

facchinm, 2019-04-09:
I'm closing the issue to keep things tidy but feel free to comment here if you have any other question/update :slightly_smiling_face:

closed 2019-04-09 by facchinm