arduino-libraries/MIDIUSB #56: arduino Due Reupload after Power Lost

joshuahornig, 2019-06-06:
I have the Problem that i have to reupload the Code every time after the Arduino Lost its power.
if i don't do that my OS dont see an USB MIDI Device
im Using an Arduino due an Upload via the Programmingport.

fab672000, 2019-06-06:
This is not a MIDIUSB problem but an arduino hardware problem, so please in future do not post here for this kind of issues.

You probably run into the power off/on known design flaw of the DUE that needs an extra reset:
See [https://copperhilltech.com/blog/arduino-due-design-flaw-due-wont-start-after-poweroffon-requires-reset/](this article here) for more information.

per1234, 2019-06-07:
Since it's off topic for this repository, I'm going to close this as invalid.

I believe we already have an issue report for this problem: https://github.com/arduino/ArduinoCore-sam/issues/22 so it's not necessary for you to report the problem in a more suitable repository.

closed 2019-06-07 by per1234