arduino-libraries/MIDIUSB #58: Not working on iOS (iphone)

samlochner, 2019-10-17:
I've got it working on a windows PC, but doesn't seem to work on iOS for me (iphone via usb to lightning adapter).

Would you expect this to work on iOS?

facchinm, 2019-10-28:
@samlochner I'm not sure you can use any USB device on iOS if it doesn't contain an MFI chip from Apple.
If it does (with some magic) you may just need to modify the core to reduce the current we ask to the host device.
Depending on the board you are using, the value should be in a place like this (https://github.com/arduino/ArduinoCore-avr/blob/master/cores/arduino/USBCore.h#L101).
Try changing 500 into 100 and please report the result here :wink:

nevvkid, 2019-10-28:
@samlochner you need a "Camera Connection Kit" for iOS devices. And your board package needs to set the power to 20mA... otherwise, it will never work. I am doing that for my product. Our pull request allows setting the power this is in the core for a long time. Here is the commit which I asked Andreas to do: https://github.com/arduino/ArduinoCore-avr/commit/6ad770e86f2758995ed8730505c9ae165ed8ef59#diff-6b60095446038cceeb7e4f1b556ea2b6 

Here is my board package as an example: https://github.com/dadamachines/automat/blob/master/software/arduino/hardware/dadmachines-automat/samd/boards.txt#L44

samlochner, 2019-10-28:
I found that if I connect the arduino usb port to a usb hub and then from the hub to the camera adapter, I don't get a power error on the iphone. But it still doesn't seem to work in garageband. Would I really get a different result if I instead changed the power instead?

I'm quite new to all this, what would I do with that pull request/commit, @nevvkid? Is that some kind of firmware that has to be flashed over? I'm using Leonardo and later a Due, btw.

nevvkid, 2019-10-28:
@samlochner Garageband doesn't work for MIDI for me at all. But not also with my products. Garageband doesn't even have MIDI anymore as far as I know. So just use Ableton to test stuff. And the USB Hub won't help either. You have to implement the setup properly and it will work. For Leonardo, there was this project with iPhone/iPad support: https://github.com/rkistner/arcore But the code is quite old now I think. Better move to a SAMD21 or SAMD51 board.

samlochner, 2019-10-28:
iOS garageband works for me only when I'm using a 5 pin midi connector via a midi to usb cable (usb side plugs into camera adapter then to iphone).

But, @nevvkid, you have been able to get midi to work over usb without 5pin connection in other iOS apps such as? And you do this by using the change? https://github.com/arduino/ArduinoCore-avr/commit/6ad770e86f2758995ed8730505c9ae165ed8ef59#diff-6b60095446038cceeb7e4f1b556ea2b6

Is that firmware that has to be flashed or something? How would I go about that?