arduino-libraries/MIDIUSB #61: Library uncompatible with the Arduino nano BLE

Vuhian, 2020-05-05:
Hello, 

First thanks for the great work with this library. I really enjoy it. 
I just noticed that the library is not compatible with the arduino nano BLE. Do you plan to make it compatible ? 
It would be great to have a midi controller able to use usb and bluetooth.

Thanks, 
Isaac

lathoub, 2020-10-04:
Working on it in https://github.com/lathoub/Arduino-USBMIDI, all help welcome :-)

algomusic, 2021-01-16:
This could also help on the Adafruit Circuit Playground Bluefruit - also currently unsupported. Thanks for your ongoing dev.

per1234, 2022-01-15:
Related: https://github.com/arduino-libraries/MIDIUSB/issues/73

amowry, 2022-06-13:
I'd also love to see this functionality. Thanks!