arduino-libraries/MIDIUSB #65: SAMD21 USBDeviceClass::ISRHandler() called continuously.

studiohsoftware, 2020-12-13:
Running the MKR1000, I discovered with a debug statement in USBDeviceClass::ISRHandler() within USBCore.cpp, the handler is called continuously. 

Executing this statement after enumeration seems to solve the problem.
USB->DEVICE.INTENCLR.reg = 0xFF;

facchinm, 2020-12-14:
Hi @studiohsoftware ,
thanks for spotting this. Would you mind creating a PR for https://github.com/arduino/ArduinoCore-samd targeting this issue? Thanks!

studiohsoftware, 2020-12-14:
I don't have a solution though. You can show the problem with a print statement in the first line within USBDeviceClass::ISRHandler(). 

Successful enumeration depends on interrupts being enabled on DEVICE.INTFLAG. I am just not sure which ones are needed, why it is firing continuously (maybe we have an interrupt enabled on DEVICE.INTFLAG unnecessarily) or when, if we need to wait until after enumeration to disable interrupts, disabling can safely occur. 

Apparently disabling everything within PluggableUSB().handleEndpoint(ep) works, but this seems like something to do in MIDIUSB.h. I am not sure the correct place to apply the fix.

So I don't have a PR ready, but I could post an issue over there. Would that help?

studiohsoftware, 2020-12-16:
Okay final thoughts after having submitted the issue to the core team. 

It seems the interrupt is happening due to the start-of-frame interrupt (SOF) and I think that the SAMD requires the user code to ACK the SOF whenever a pipe of type interrupt or iso is in use. So disabling that interrupt prevents the ACK programmed within USBDeviceClass::ISRHandler(), and enumeration on ep 0 leads to a stall condition. 

Since MIDIUSB does not use interrupt or iso ep's, it means that this library unnecessarily visits USBDeviceClass::ISRHandler() every 1ms because of the SOF interrupt. It would be nice to have this interrupt disabled after enumeration, if possible. It seems that the interrupt can safely disabled once PluggableUSB.handleEndpoint. 

In other words, this request comes down to implementing handleEndpoint within MIDIUSB.h and disabling the interrupt there.