arduino-libraries/MIDIUSB #94: casting error in COntrolChange and sendnote

lorenzopetralia, 2023-01-16:


per1234, 2023-01-17:
Hi @lorenzopetralia. It is important to understand the difference between warnings and errors. A warning is the compiler telling you there is something in the code that could possibly cause a problem but doesn't cause the compilation to fail. An error is a problem with the code that causes compilation to fail. In this case you have posted warnings, not errors.

>Compiling does not work anymore

This is incorrect. The output you shared shows a successful compilation. So you can get back to progressing with your project. If you require any assistance working on your project or fixing the warnings in your sketch code, you can post on the Arduino Forum:

https://forum.arduino.cc/c/using-arduino/audio/24

I'm sure we will be able to help you out over there.

per1234, 2023-01-17:
Closing as duplicate of https://github.com/arduino-libraries/MIDIUSB/issues/91

closed 2023-01-17 by per1234