arduino-libraries/MIDIUSB #97: Disconnections and wrong behavior with music applications

Mark-81, 2023-05-29:
I wrote a code based on the examples to read three ADC inputs and transmit MIDI messages to the PC. I'm using a custom Leonardo board.
Here my code:

```
#define ADC_BASE              18

typedef struct
{
    int value;
    int shadow;
} slider_t;

slider_t sliders[3];
byte cc[3] = {1, 11, 2};

void loop() 
{
  for (byte i = 0; i < 3; i++) sendMIDI(i, analogRead(ADC_BASE + i));
}

void sendMIDI(byte idx, int adc)
{  
    sliders[idx].value = map(adc, 0, 1023, 0, 127);
    if (sliders[idx].shadow != sliders[idx].value)
    {
        if (MidiUSB.isAvailable() == 0) sendControlChange(0, cc[idx], sliders[idx].value);
        sliders[idx].shadow = sliders[idx].value;
    }
}

void sendControlChange(byte channel, byte control, byte value) 
{
    midiEventPacket_t event = { EVENT_CONTROL_CHANGE, (byte) (0xB0 | channel), control, value };
    MidiUSB.sendMIDI(event);
    MidiUSB.flush();
}
```

On Linux `aseqdump` receives the messages with no problems for 8+ hours.
On Windows 10 I'm facing these issues:

**Cubase v12.0.60**
After few seconds, Cubase does not receive anymore the messages. I have to disconnect, close the application and restart it.
Sometimes, instead, the GUI of Cubase freezes but it is still alive - it continues to play music.

**Kontakt Full Standalone v7.3.0**
The application receives the messages, but suddenly the virtual keyboard in Kontakt loses the blue color (like if the instrument was disabled). I have to re-assign the instrument to the channel to play. But again, after moving my slider and sending a MIDI message, the application drop the instrument.

Is there something wrong in my code?
Has anybody any experience in a real case scenario?

Mark-81, 2023-05-30:
I tried to add a small delay after a write, i.e.:

```
void sendControlChange(byte channel, byte control, byte value) 
{
    midiEventPacket_t event = { EVENT_CONTROL_CHANGE, (byte) (0xB0 | channel), control, value };
    MidiUSB.sendMIDI(event);
    MidiUSB.flush();
    delay(10);
}
```

but the behavior is always the same.

Mark-81, 2023-06-14:
**UPDATE**

I noticed after some time the channel is no more writable (`USB_Available(MIDI_TX);` returns 0) no matter how many packets I sent before. It may happens after few seconds or minutes. But this happens on Windows _only_. Linux and Mac work fine. I used different MIDI monitors and the behavior is always the same. Hence is not a problem with those applications.

I tried on four machines, with Windows 10 and 11. I changed the USB cable and tested three Leonardo boards.
When it happens a reset of the board does not recover the functionality. I need to physically disconnect and reconnect the USB cable.

I also added a `MidiUSB.read();` in the main loop in order to ignore any incoming packets.

Any suggestion? I'm pretty sure for others it works fine in Windows!

GuitarPikstar, 2024-08-08:
Did you find any solution to this problem. We face the exact same phenomena with cubase and we use it for live performance, which is kinda critical...