arduino-libraries/MIDIUSB #98: Support mbed/stm32 devices like Arduino GIGA

Haschtl, 2023-07-17:
This library currently only supports these architectures: avr, sam, samd
Is there any plan to support the newer STM32 boards like the Portenta boards or Arduino GIGA?

I recently bought the Arduino GIGA and justs expected this library to be supported.

per1234, 2023-07-17:
Hi @Haschtl. Thanks for your suggestion.

I see a "USBMIDI" library has been added to the `arduino/ArduinoCore-mbed` repository that contains the codebase of the Giga's "**Arduino Mbed OS Giga Boards**" platform:

https://github.com/arduino/ArduinoCore-mbed/commit/35a9e18b5d27b67f190f2626425326a66abdd58a

Unfortunately even though the library was added to the repository, the platform packaging system has not been configured to include that library in the distributed "**Arduino Mbed OS Giga Boards**" platform. I'm not sure if that was an oversight, or done intentionally because the library is not yet ready for distribution. You could download the library files from GitHub and install them manually if you want to give it a try. They are here:

https://github.com/arduino/ArduinoCore-mbed/tree/main/libraries/USBMIDI

---

Related:

- https://github.com/arduino-libraries/MIDIUSB/issues/61
- https://github.com/arduino-libraries/MIDIUSB/issues/73

Haschtl, 2023-07-18:
Thanks for the clarification!

I copied the [USBMIDI](https://github.com/arduino/ArduinoCore-mbed/tree/main/libraries/USBMIDI)-folder from the ArduinoCore-mbed repo to my arduino library (`~/.arduino15/packages/arduino/hardware/mbed_giga/4.0.2/libraries`) and compiled+uploaded the  included example `Basic.ino` without problems. The device is detected as a MIDI-Device now!

Haschtl, 2023-07-18:
The underlying USBMIDI library for GIGA is working, but `MIDIUSB` does not use it yet.
I had a look at the source code and tried to include the `USBMIDI` library from the mbed_giga package without success.
I thought adding something like this to `MIDIUSB.h` would be enough:
```
#elif defined(__MBED__)

#include "PluggableUSBMIDI.h"

#define EPTYPE_DESCRIPTOR_SIZE uint8_t
#define EP_TYPE_BULK_IN_MIDI EP_TYPE_BULK_IN
#define EP_TYPE_BULK_OUT_MIDI EP_TYPE_BULK_OUT
#define MIDI_BUFFER_SIZE USB_EP_SIZE
#define is_write_enabled(x) (1)
```
But there are still some undefined references (`USBCON`, `EndpointDescriptor`, `IADDescriptor`, `InterfaceDescriptor`, `USB_EP_SIZE`, `D_IAD`, `D_INTERFACE`, `USB_ENDPOINT_OUT`, `USB_ENDPOINT_IN`, `EP_TYPE_BULK_OUT`, `EP_TYPE_BULK_IN`)

I don't know if this approach is in any case correct...

My goal is to get [lathoub's USBMIDI](https://github.com/lathoub/Arduino-USBMIDI) running. If `MIDIUSB` works with the GIGA, can I expect that lathoub's library will work, too?

stfallet, 2025-04-25:
Hi, were you able to resolve the reading issue? I'm struggling to get my Giga to work with USBMIDI. thx for any advice.