arduino-libraries/MIDIUSB #99: Arduino Zero MIDIUSB problem

ccarmatic, 2023-08-11:
I am using Windows 11 and an Arduino Zero

this is my code

```
#include <Arduino.h>
#include "MIDIUSB.h"

unsigned long prevtime = 0;
unsigned long time = 0;
int prevvalue = 0;

void controlChange(byte channel, byte control, byte value)
{
  midiEventPacket_t event = {0x0B, 0xB0 | channel, control, value};
  MidiUSB.sendMIDI(event);
  // MidiUSB.flush();
}

void setup()
{
  prevvalue = analogRead(A0) / 8;
}

void loop()
{

  digitalWrite(LED_BUILTIN, LOW);
  time = millis() - prevtime;
  if (time > 50)
  {
    digitalWrite(LED_BUILTIN, HIGH);
    uint8_t value = analogRead(A0) / 8;
    if (abs(value - prevvalue) > 5)
    {
      prevvalue = value;
      controlChange(1, 1, value);
      MidiUSB.flush();
    }
    
    prevtime = millis();
  }
}
```

This code reads the analog value from A0 and sends it as a MIDI value on control 1, channel 1 , if the analog reading has changed above a threshold 
and it repeats the process every 50 milliseconds

I flash the code to the Zero using its debug USB port and everything worked, and I can verify that it is working because the builtin LED flashes correctly and the transmit LED lights up when I turn the trimpot connected to A0

But when I connect the Zero with the native port (which MIDIUSB uses) , I see this in Windows Device Manager:
[image]
an entry with the Arduino Zero name at COM7 which 'does not exist' to Windows, and a 'USB Serial Device' at COM6 

**The Arduino Zero is not being detected as a MIDI device** , but Arduino IDE can access COM6 to upload sketches and open the serial monitor on the native USB port

If I upload any other sketch , which does not contain MIDIUSB, everything is normal , in Windows, in Arduino IDE, and on the Arduino Nano
**This leads me to believe that the fault lies in MIDIUSB**

per1234, 2023-08-11:
@ccarmatic's post about the problem on Arduino Forum:

https://forum.arduino.cc/t/arduino-zero-midiusb-problem/1157189

Related:

- https://github.com/arduino-libraries/MIDIUSB/issues/60
  - Note the reply:
    > if you are on Windows, it may happen that the drivers for the new composite device are not getting loaded correctly. Try uninstalling all the drivers currently loaded with the board (from Device Manager) and if you are on windows 10 it should reinstall the correct ones.
For further explanation, please read this thread that almost completely applies (https://github.com/webusb/arduino/issues/44)
- https://github.com/arduino-libraries/MIDIUSB/issues/2

ccarmatic, 2023-08-11:
Hello, I have tried the uninstall and reinstall drivers method without success - until I closed the Arduino IDE and did it again - the IDE was preventing Windows from deleting the necessary drivers