arduino-libraries/MIDIUSB #102: NoteOn and Off in clock-example

JlnWntr, 2023-09-23:
Why do you send these notes?

https://github.com/arduino-libraries/MIDIUSB/blob/f9bcc4eaf7bb983ac1827ccf0f8b833f75001ac2/examples/MIDIUSB_clock/MIDIUSB_clock.ino#L49

https://github.com/arduino-libraries/MIDIUSB/blob/f9bcc4eaf7bb983ac1827ccf0f8b833f75001ac2/examples/MIDIUSB_clock/MIDIUSB_clock.ino#L56

https://github.com/arduino-libraries/MIDIUSB/blob/f9bcc4eaf7bb983ac1827ccf0f8b833f75001ac2/examples/MIDIUSB_clock/MIDIUSB_clock.ino#L62

facchinm, 2023-11-03:
That's likely a question for @ErniW who contributed to that example back in 2016 :slightly_smiling_face:

fab672000, 2023-11-03:
> Why do you send these notes?
> 
Its just for debugging purpose to know it works, the (same) note is sent every ppqm interval of 24  (minimal standard Sync24 resolution) so every quarter notes of a received clock and also on Clock start/stop events (0xFA and 0xFC). 
If the arduino receiver actually works, you should be able to hear it thanks to that notes emitted there to your synth/sound module device.

ErniW, 2023-11-06:
@JlnWntr Well, actually @fab672000 answered your question. The purpose to place them was to check bi-directional communication and how does Arduino respond on start/stop messages without reading them manually. If you don't need it you can remove them.

closed 2023-11-06 by facchinm