arduino-libraries/MIDIUSB #114: Due USB Name change

RezaYazdi604, 2024-05-19:
Hi

I'm using this library for making a MIDI controller with Arduino Due. 
I want to change the name of the native USB port on my Due to show a custom name when connected to PC.
Does anyone here know how to do this for Due? 

I've seen it that they can do it for other AVR boards but nor for Due. 

Appreciate any help and guidance.
Thanks

per1234, 2024-05-19:
Hi @RezaYazdi604. Thanks for your interest in this open source project. This issue tracker is only to be used to report bugs or feature requests specific to the project. This topic is more appropriate for the Arduino Forum. I'm sure we will be able to help you out over there:

https://forum.arduino.cc/

closed 2024-05-19 by per1234

RezaYazdi604, 2024-05-19:
Thank you. I’ve posted my problem on that forum too. Hopefully someone can help. If you know any previous posts to address my problem please point me there. Thanks again.  Cheers,Reza

https://forum.arduino.cc/